neufieldrobotics/spinnaker_sdk_camera_driver #129: Cannot launch node as normal user, but can launch as root

civerachb-cpr, 2021-06-03:
**System Description**
Camera Name and Model: 2x Blackfly S (USB3, connected via Sync cable)
Operating System: Ubuntu 18.04.5 w/ JetPack 4.5.1
Spinnaker Version: 2.2.0.48
ROS Version: Melodic

***Computer details***
Processor: Nvidia AGX Xavier
RAM: 32GB

***Multiple Cameras Only***
How is the triggering setup? Master/Slave  (though not specifically applicable to my current problem -- see below)

For USB 3.0 cameras:
On same USB 3.0 bus or different buses? Same bus
Is the USB 3.0 hub externally powered? Yes

**Do the cameras work with SpinView?**
Yes

**Describe the bug**
Summary: I cannot launch the node/nodelet as a normal user, but I can launch it correctly if I am logged into a root shell.

Following the instructions in https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/issues/89 I am able to build the node.  However, when I try to launch the node I get the commonly-cited `[FATAL] [1622735926.930668531]: No cameras found!` failed assertion.  But, if I run the following:

```
$ sudo -s
# source /opt/ros/melodic/setup.bash
# source /home/administrator/catkin_ws/devel/setup.bash
# roslaunch roslaunch spinnaker_sdk_camera_driver node_acquisition.launch binning:=2
```

the note starts up properly and I am able to see data; `rostopic hz` reports the data is coming in at about 20Hz and I can see the video stream in rviz.

For simplicity I'm only trying to launch a single camera for the moment.  Once I get these issues sorted out I'll re-enable the synchronization and launch the second camera.

I suspect that it's something as simple as a udev rule that I'm missing, but I'm not sure what the necessary rule is.  I have the following rule defined in `/etc/udev/rules.d/50-spinnaker.rules`:

```
SUBSYSTEM=="usb", ATTR{idVendor}=="1e10", MODE:="0666", GROUP="flirimaging", SYMLINK+="spinnaker%n
```

Checking the normal user's groups:

```
$ groups
administrator adm cdrom sudo audio dip video plugdev netdev i2c lpadmin gdm lightdm sambashare weston-launch gpio flirimaging
```

I can see both camera devices:

```
$ ls -l /dev/spinnaker*
lrwxrwxrwx 1 root root 15 Jun  2 17:35 /dev/spinnaker1 -> bus/usb/002/033
lrwxrwxrwx 1 root root 15 Jun  2 17:35 /dev/spinnaker4 -> bus/usb/002/034

$ ls -l /dev/bus/usb/002/033
crw-rw-rw- 1 root flirimaging 189, 160 Jun  3 12:07 /dev/bus/usb/002/033

$ ls -l /dev/bus/usb/002/034
crw-rw-rw- 1 root flirimaging 189, 161 Jun  3 12:08 /dev/bus/usb/002/034
```

Is this a known limitation of the driver and/or libspinnaker that it must be launched as root?  Or is this something specific to the AGX Xavier and/or the version of JetPack that I'm using?  Or is there an additional group that I need to add the user to that I've omitted?

**Error Messages**
Running from a root shell:

```
$ sudo -s
[... 224 lines cut ...]
[ INFO] [1622736523.756362628]:    -0.021142 -0.373387 2.385983 3.282457 
[ INFO] [1622736523.760070166]:   Camera Rectification Paramters:
[ INFO] [1622736523.760438536]:    1.000000 0.000000 0.000000 0.000000 1.000000 0.000000 0.000000 0.000000 1.000000 
[ INFO] [1622736523.764589840]:   Camera Projection Paramters:
[ INFO] [1622736523.764937281]:    913.700317 0.000000 953.448302 0.000000 0.000000 1063.296631 777.871993 0.000000 0.000000 0.000000 1.000000 0.000000 
[ INFO] [1622736523.765054758]:   Camera coeffs provided, camera info messges will be published.
[ INFO] [1622736523.765434937]: *** SYSTEM INFORMATION ***
[ INFO] [1622736523.765602881]: Creating system instance...
[ INFO] [1622736523.765780041]: spinnaker_sdk_camera_driver package version: 1.1.0
[ INFO] [1622736525.122585618]: Spinnaker library version: 2.2.0.48
[ INFO] [1622736525.123166638]: Retreiving list of cameras...
[FATAL] [1622736525.128492366]: ASSERTION FAILED
	file = /home/administrator/catkin_ws/src/spinnaker_sdk_camera_driver/src/capture.cpp
	line = 185
	cond = numCameras_
	message = 
[FATAL] [1622736525.128690040]: No cameras found!
[FATAL] [1622736525.128794973]: 

[acquisition_node-2] process has died [pid 6482, exit code -6, cmd /home/administrator/catkin_ws/devel/lib/spinnaker_sdk_camera_driver/acquisition_node __name:=acquisition_node __log:=/home/administrator/.ros/log/f70e1e5e-c485-11eb-bce2-00044bcbdf38/acquisition_node-2.log].
log file: /home/administrator/.ros/log/f70e1e5e-c485-11eb-bce2-00044bcbdf38/acquisition_node-2*.log
```

The referenced log file does not exist, but here's the roslaunch log from the same directory:
[roslaunch-cpr-mitre-6442.log](https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/files/6592971/roslaunch-cpr-mitre-6442.log)

vik748, 2021-06-03:
Usually the last step in installing the Spinnaker library adds the current user to the right groups with permissions to the usb etc.  Are you sure you have installed the library correctly? Are you able to run the camera using SpinView as a standard user?

civerachb-cpr, 2021-06-04:
Yes, I'm able to use SpinView as a normal user.

After rebooting the Jetson a few times it looks like the issue may have resolved itself for reasons I haven't yet identified.  Re-running `groups` shows the same output as previously, so nothing's changed on that front.  To install the Spinnaker library + tools I just downloaded the archive, unzipped it, and installed with `sudo dpkg -i` and `sudo apt-get -f install` to grab a few missing dependencies.

I don't have a good explanation for _why_ it seems to be working now, but I'll keep an eye on it and leave another comment if it suddenly stops for some reason.

vik748, 2021-06-04:
Ideally you want to use the `install_spinnaker_arm.sh` script included with Spinnaker.

vik748, 2021-06-14:
@civerachb-cpr  I am closing the issue since we haven't heard from you.  Pls re-open if necessary.

closed 2021-06-14 by vik748

civerachb-cpr, 2021-06-21:
Sorry about the radio silence.  I was unable to replicate the original problem.  Not sure what exactly happened, but it appears to have resolved itself, so it's all good now.  Thanks!