neufieldrobotics/spinnaker_sdk_camera_driver #135: Unable to execute trigger. Aborting...

Karthikeyanc2, 2021-06-16:
Hi,

Thanks for the nice tool and the support. 

I have followed your previous answers to comment out the configurations, which cameras does not allow. But I faced some other issued which I am not  able to understand.

Specs:
Ubuntu 18.04
Spinnaker 2.4.0
Camera: Flir ADK with Ethernet

Terminal output:
linux2@linux2-HP-ProBook-440-G5: ~/flir_ws$ source devel/setup.bash 
linux2@linux2-HP-ProBook-440-G5: ~/flir_ws$ roslaunch spinnaker_sdk_camera_driver acquisition.launch
... logging to /home/linux2/.ros/log/4e039a22-ce9a-11eb-8ff6-b4b686e9bfb1/roslaunch-linux2-HP-ProBook-440-G5-2888.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://linux2-HP-ProBook-440-G5:39705/

SUMMARY

PARAMETERS
 * /acquisition_node/binning: 1
 * /acquisition_node/cam_aliases: ['cam0']
 * /acquisition_node/cam_ids: [19426159]
 * /acquisition_node/color: False
 * /acquisition_node/delay: 1.0
 * /acquisition_node/distortion_coeffs: [[-0.021141875266...
 * /acquisition_node/distortion_model: plumb_bob
 * /acquisition_node/exposure_time: 0
 * /acquisition_node/external_trigger: False
 * /acquisition_node/flip_horizontal: [False]
 * /acquisition_node/flip_vertical: [False]
 * /acquisition_node/frames: 3400
 * /acquisition_node/gain: 0
 * /acquisition_node/image_height: 512
 * /acquisition_node/image_width: 640
 * /acquisition_node/intrinsic_coeffs: [[1886.9232141485...
 * /acquisition_node/live: False
 * /acquisition_node/live_grid: False
 * /acquisition_node/master_cam: 19426159
 * /acquisition_node/max_rate_save: False
 * /acquisition_node/projection_coeffs: [[913.700317, 0.0...
 * /acquisition_node/rectification_coeffs: [[1.0, 0.0, 0.0, ...
 * /acquisition_node/region_of_interest/height: 0
 * /acquisition_node/region_of_interest/width: 0
 * /acquisition_node/region_of_interest/x_offset: 0
 * /acquisition_node/region_of_interest/y_offset: 0
 * /acquisition_node/save: False
 * /acquisition_node/save_path: ~
 * /acquisition_node/save_type: bmp
 * /acquisition_node/skip: 20
 * /acquisition_node/soft_framerate: 30
 * /acquisition_node/target_grey_value: 0
 * /acquisition_node/tf_prefix: 
 * /acquisition_node/time: False
 * /acquisition_node/to_ros: True
 * /acquisition_node/utstamps: False
 * /rosdistro: melodic
 * /rosversion: 1.14.11

NODES
  /
    acquisition_node (nodelet/nodelet)
    vision_nodelet_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [2905]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 4e039a22-ce9a-11eb-8ff6-b4b686e9bfb1
process[rosout-1]: started with pid [2930]
started core service [/rosout]
process[vision_nodelet_manager-2]: started with pid [2933]
process[acquisition_node-3]: started with pid [2938]
[ERROR] [1623844770.318092447]: Skipped loading plugin with error: XML Document '/home/linux2/lanelet2_ws/src/sim_sample_actuator_ros_tool/nodelet_plugins.xml' has no Root Element. This likely means the XML is malformed or missing..
[ INFO] [1623844770.324042221]: Initializing nodelet with 8 worker threads.
[ INFO] [1623844770.590531232]: Initializing nodelet
[ INFO] [1623844770.592661290]: [ OK ] USB memory: 1000 MB
[ INFO] [1623844770.592761357]: *** PARAMETER SETTINGS ***
[ INFO] [1623844770.592821190]: ** Date = 20210616
[ INFO] [1623844770.593428448]:   Save path set via parameter to: /home/linux2
[ INFO] [1623844770.593503793]:   Camera IDs:
[ INFO] [1623844770.593994790]:     19426159
[ INFO] [1623844770.594399051]:   Camera Aliases:
[ INFO] [1623844770.594450358]:     19426159 >> cam0
[ INFO] [1623844770.594795693]:   External trigger: false
[ INFO] [1623844770.595565038]:   Unique time stamps for each camera: false
[ INFO] [1623844770.595945332]:   color set to: false
[ INFO] [1623844770.596389324]:   19426159 flip_horizontal 0
[ INFO] [1623844770.596922063]:   19426159 flip_vertical 0
[ INFO] [1623844770.597352254]:   Exporting images to ROS: true
[ INFO] [1623844770.597774707]:   Showing live images setting: false
[ INFO] [1623844770.598206404]:   Showing grid-style live images setting: false
[ INFO] [1623844770.598640551]:   Max Rate Save Mode: false
[ INFO] [1623844770.599086475]:   Displaying timing details: false
[ INFO] [1623844770.599515159]:   No. of images to skip set to: 20
[ INFO] [1623844770.599970292]:   Init sleep delays set to : 1.00 sec
[ WARN] [1623844770.600340921]:   'fps' Parameter not set, using default behavior: fps=20.00
[ INFO] [1623844770.600699402]:   'exposure_time'=0, Setting autoexposure
[ INFO] [1623844770.601060909]:   'gain' Parameter was zero or negative, using Auto gain based on target grey value
[ INFO] [1623844770.601419714]:   'target_grey_value'=0, Setting AutoExposureTargetGreyValueAuto to Continuous/ auto
[ INFO] [1623844770.601790320]:   Binning set to: 1
[ INFO] [1623844770.602231765]:   Using Software rate control, rate set to: 30
[ INFO] [1623844770.602808508]:   Saving images set to: 0
[ INFO] [1623844770.604175803]:   tf_prefix set to: 
[ INFO] [1623844770.606293903]:   Region of Interest set to width: 0	height: 0	offset_x: 0 offset_y: 0
[ INFO] [1623844770.606730102]:   Camera Intrinsic Paramters:
[ INFO] [1623844770.607056932]:    1886.923214 0.000000 604.721488 0.000000 1886.666877 493.477267 0.000000 0.000000 1.000000 
[ INFO] [1623844770.607470001]:   Camera Distortion Paramters:
[ INFO] [1623844770.607521562]:    -0.021142 -0.373387 2.385983 3.282457 
[ INFO] [1623844770.607976738]:   Camera Rectification Paramters:
[ INFO] [1623844770.608021722]:    1.000000 0.000000 0.000000 0.000000 1.000000 0.000000 0.000000 0.000000 1.000000 
[ INFO] [1623844770.608495400]:   Camera Projection Paramters:
[ INFO] [1623844770.608546677]:    913.700317 0.000000 953.448302 0.000000 0.000000 1063.296631 777.871993 0.000000 0.000000 0.000000 1.000000 0.000000 
[ INFO] [1623844770.608567158]:   Camera coeffs provided, camera info messges will be published.
[ INFO] [1623844770.608657991]: *** SYSTEM INFORMATION ***
[ INFO] [1623844770.608682624]: Creating system instance...
[ INFO] [1623844770.608745521]: spinnaker_sdk_camera_driver package version: 1.1.0
[ INFO] [1623844771.981337557]: Spinnaker library version: 2.4.0.143
[ INFO] [1623844771.981467760]: Retreiving list of cameras...
[ INFO] [1623844771.988963681]: Numer of cameras found: 1
[ INFO] [1623844771.989083259]:  Cameras connected: 1
[ INFO] [1623844771.991154264]:   -19426159 Boson-GE-640 1901.1.325.0
[ INFO] [1623844772.060471468]: Dynamic Reconfigure: Level : 4294967295
[ INFO] [1623844772.061675776]: *** FLUSH SEQUENCE ***
[ INFO] [1623844772.061706843]: Initializing cameras...
[ INFO] [1623844772.902820685]: Deinitializing cameras...
[ INFO] [1623844773.113435864]: All cameras deinitialized.
[ INFO] [1623844775.113899323]: Initializing cameras...
[ INFO] [1623844775.638678487]: onInit Initialized
[ INFO] [1623844775.638911396]: *** ACQUISITION ***
[FATAL] [1623844775.802457724]: Unable to execute trigger. Aborting...
terminate called after throwing an instance of 'Spinnaker::Exception'
  what():  Spinnaker: GenICam::AccessException= Node is not writable. : AccessException thrown in node 'TriggerSoftware' while calling 'TriggerSoftware.Execute()' (file 'CommandT.h', line 61) [-2006]
[vision_nodelet_manager-2] process has died [pid 2933, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=vision_nodelet_manager __log:=/home/linux2/.ros/log/4e039a22-ce9a-11eb-8ff6-b4b686e9bfb1/vision_nodelet_manager-2.log].
log file: /home/linux2/.ros/log/4e039a22-ce9a-11eb-8ff6-b4b686e9bfb1/vision_nodelet_manager-2*.log

I hope the solution is trivial, but I am not seeing it.
Thanks for the help,
Karthik

vik748, 2021-06-16:
@Karthikeyanc2 we use software triggering in this driver to ensure that none of the frames go into the buffer and cause lag. I looks like your camera might not support software triggering.  Can you try to create a similar configuration in SpinView and check if it lets you do a software trigger on the camera.

Karthikeyanc2, 2021-06-16:
Hi @vik748 ,

Thanks for the quick reply,

Yes, with spinview, it works with the setting: "Trigger source : software"

[image]

vik748, 2021-06-16:
After you switch to Trigger source software, does the live image from the camera freeze?  Does it then take a picture when you click 'Execute' for the software trigger?
Also I noticed your Spinview is 2.3.xx, if you see the README there are known issues with Spinnaker versions above 2.2, can you try one of the tested versions first.

Karthikeyanc2, 2021-06-16:
I just tried with spinnaker 2.0.0, 

[image]

its exactly the same issue, 

[image]

at spinview, clicking on 'Execute' has no effect. (Also, on whichever trigger source I select and whatever I do with regards to this option, it has no effect, frames are live and does not freeze).

Also, I could not find spinnaker 2.2 in this download website, https://www.flir.eu/products/spinnaker-sdk/. Do you have any other download source?

vik748, 2021-06-16:
I believe that once you set the camera to software trigger, it should stop triggering and the image should freeze, even if you high the stop and play button. I don't have a camera with me to try this.
It is possible that this camera doesn't support software triggering.  In that case you'd have to modify the code to make it similar to what it does in the MAX_RATE_SAVE mode at https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/blob/f6ab2909ba2a4e250d7838331c83d4b588badb0f/src/capture.cpp#L740

You'd also have to remote the trigger execution line in the main loop.