neufieldrobotics/spinnaker_sdk_camera_driver #137: Cannot launch node of type [spinnaker_sdk_camera_driver/acquisition_node]

anoneuro, 2021-06-22:
System: Ubuntu 16.04
ROS distro: Kinetic
Camera: Blackfly S USB3

SpinView works perfectly.

**Here is the error I'm seeing:**

... logging to /home/rancher/.ros/log/a219d8e6-d364-11eb-a253-d0c637d41db3/roslaunch-rancher-NUC8i7BEH-9580.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://rancher-NUC8i7BEH:44049/

SUMMARY
========

PARAMETERS
 * /acquisition_node/binning: 1
 * /acquisition_node/cam_aliases: ['cam0']
 * /acquisition_node/cam_ids: [20539276]
 * /acquisition_node/color: False
 * /acquisition_node/delay: 1.0
 * /acquisition_node/distortion_coeffs: [[-0.021141875266...
 * /acquisition_node/distortion_model: plumb_bob
 * /acquisition_node/exposure_time: 0
 * /acquisition_node/external_trigger: False
 * /acquisition_node/flip_horizontal: [False]
 * /acquisition_node/flip_vertical: [False]
 * /acquisition_node/frames: 3400
 * /acquisition_node/image_height: 1080
 * /acquisition_node/image_width: 1440
 * /acquisition_node/intrinsic_coeffs: [[1886.9232141485...
 * /acquisition_node/live: False
 * /acquisition_node/live_grid: False
 * /acquisition_node/master_cam: 20539276
 * /acquisition_node/max_rate_save: False
 * /acquisition_node/projection_coeffs: [[913.700317, 0.0...
 * /acquisition_node/rectification_coeffs: [[1.0, 0.0, 0.0, ...
 * /acquisition_node/region_of_interest/height: 0
 * /acquisition_node/region_of_interest/width: 0
 * /acquisition_node/region_of_interest/x_offset: 0
 * /acquisition_node/region_of_interest/y_offset: 0
 * /acquisition_node/save: False
 * /acquisition_node/save_path: ~
 * /acquisition_node/save_type: bmp
 * /acquisition_node/skip: 20
 * /acquisition_node/soft_framerate: 20
 * /acquisition_node/target_grey_value: 0
 * /acquisition_node/tf_prefix: 
 * /acquisition_node/time: False
 * /acquisition_node/to_ros: True
 * /acquisition_node/utstamps: False
 * /rosdistro: kinetic
 * /rosversion: 1.12.17

NODES
  /
    acquisition_node (spinnaker_sdk_camera_driver/acquisition_node)

ROS_MASTER_URI=http://localhost:11311

ERROR: cannot launch node of type [spinnaker_sdk_camera_driver/acquisition_node]: can't locate node [acquisition_node] in package [spinnaker_sdk_camera_driver]
No processes to monitor
shutting down processing monitor...
... shutting down processing monitor complete

Any help on this would be greatly appreciated.
Thanks in advance!

vik748, 2021-06-22:
Are you sure the workspace is sourced correctly?

anoneuro, 2021-06-22:
Interesting. I thought I had, but just retried it and it worked perfectly. Thanks!

closed 2021-06-22 by anoneuro