neufieldrobotics/spinnaker_sdk_camera_driver #138: Camera is on a wrong subnet. [-1015]

Xulei-Zhai, 2021-06-24:
Last time,I have run this project on Ubuntu 20.04, but i was failed. You suggested me to use the tested spinnaker  version. This time , I used the tested spinnaker version 2.2.0.48 on ubuntu18.04， and meet the probelems as flollows.
[subnet-wrong.txt](https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/files/6708820/subnet-wrong.txt)

**System Description**
Camera Name and Model:FLIR  & ZX-SF1220C
Operating System:Ubuntu 18.04
Spinnaker Version:2.2.0.48
ROS Version: 1.14.11

[FATAL] [1624533412.635966106]: Error: Spinnaker: Camera is on a wrong subnet. [-1015]
[ WARN] [1624533412.635987315]: Most likely cause for this error is if your camera can't support color and your are trying to set it to color mode
[FATAL] [1624533412.816388305]: Failed to load nodelet '/acquisition_node` of type `acquisition/Capture` to manager `vision_nodelet_manager'
terminate called after throwing an instance of 'Spinnaker::Exception'
  what():  Spinnaker: Could not stop acquisition. Access denied [-1005]
[vision_nodelet_manager-2] process has died [pid 3128, exit code -6, cmd /opt/ros/melodic/lib/nodelet/nodelet manager __name:=vision_nodelet_manager __log:=/home/good/.ros/log/ac6bb07e-d4dd-11eb-80e3-a40c66050bb4/vision_nodelet_manager-2.log].
log file: /home/good/.ros/log/ac6bb07e-d4dd-11eb-80e3-a40c66050bb4/vision_nodelet_manager-2*.log
[acquisition_node-3] process has died [pid 3129, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load acquisition/Capture vision_nodelet_manager __name:=acquisition_node __log:=/home/good/.ros/log/ac6bb07e-d4dd-11eb-80e3-a40c66050bb4/acquisition_node-3.log].
log file: /home/good/.ros/log/ac6bb07e-d4dd-11eb-80e3-a40c66050bb4/acquisition_node-3*.log

vik748, 2021-06-24:
@Xulei-Zhai are you able to run the camera with SpinView?

Xulei-Zhai, 2021-06-25:
@vik748 
Thanks for your reply.
I didn't use spinview before you told me. I just opened spinview on the terminal, and the interface shows that an adapter is connected. But when I click the start device acquisition button, the terminal shows device not open. I do not have a detailed manual for spinview and still can't solve the device not open problem.

Xulei-Zhai, 2021-06-25:
@vik748 
When I restart the computer and run the specified launch file. 
First, the terminal will report an error ("first frame ID was not zero!..."). 
[image]

Then, when I run the project once again, the terminal will prompt the following error every time ("GenICam::OutOfRangeException=Value = 1 must be equal or greater than Min = 2...")
[image]

Looking forward for your reply. Thank you very much.

vik748, 2021-06-26:
We can't really help till you are able to make this work with SpinView or Acquisition examples provided with the SDK. You can to connect with FLIR support for that.
Once you get your system to talk to the camera I can help debug issues with the ROS driver.

Xulei-Zhai, 2021-09-03:
It can now work on spinview, and I can video through spinview. However, when I run ROS driver, it still raises problems as shown in the above two pictures. We look forward to your reply.

vik748, 2021-09-03:
Have you tried setting binning to 2 as suggested in the error message? You can set it in the `.launch` file.

Xulei-Zhai, 2021-09-03:
yes，We have set it to 2 in the launch file, and there is a problem in the second picture.

vik748, 2021-09-03:
Which version of spinnaker sdk and ROS are you using?

Xulei-Zhai, 2021-09-03:
spinaker ：1.24.0.60
ros： melodic

Xulei-Zhai, 2021-09-03:
I tried spinnaker version 2.2.0.48 before（i know it was tested on ubuntu18.04）, but there is still a problem

vik748, 2021-09-03:
I would suggest sticking to 2.2.0.48 for melodic since I know that combination works. Unfortunately I have not ever seen or used the camera that you are using, so not sure what to say.  
All I can suggest is that you try to look at the parameters in SpinView and try to set them to the settings we are using in the ROS driver - https://github.com/neufieldrobotics/spinnaker_sdk_camera_driver/blob/9c9633038c17d06f4ba11da9a97dddeb59c0d72a/src/capture.cpp#L672

Hopefully that can help you identify which configuration is causing the issue.

Xulei-Zhai, 2021-09-03:
Thank you very much. 
[image]
[image]
[image]
[image]
[image]
I have just successfully run this package, which mainly includes two steps. The first step is to change the value of binding in the corresponding launch file to 2. The second step is to comment out the code in line 59 mentioned in the error message. Then it runs successfully.

As shown in the figure below, thank you again for your help.

closed 2021-09-06 by Xulei-Zhai