neufieldrobotics/spinnaker_sdk_camera_driver #144: Software triggering of BFS camera's using timestamp value

ajay1606, 2021-09-23:
Hello, 

Will it be possible to trigger the PTP enabled camera (BFS model) with a specific timestamp value?  
For example, Suppose the camera has been synchronized with the PTP clock source, and through the software trigger, will it be possible to set a specific time value to trigger the camera. ?

Any response is much appreciated.

Regards,
Ajay

mithundiddi, 2021-11-15:
By PTP enabled cameras, did you mean to say your linux machine/computer is using PTP to synchronize time with local server, and you connected Blackfly S (usb) camera to your computer. Pl. let me know if I understood it wrong.

Our package doesn't natively support triggering at specific time through software trigger functionality. I would suggest using external/hardware trigger mode of our package, it does involve adding hw to generate trigger pulse at desired timestamp.

ajay1606, 2021-11-16:
@mithundiddi Thanks for your response. We are using an ethernet camera (GigE) and connected to Linux machine using LAN not USB. And rest of your understanding is clear. Few of camera models I have worked before, has option to trigger the camera through time stamp value,in case its synchronized with PTP source. Hence wonder, wether this ROS driver supports same with BF cameras .

closed 2023-03-21 by ajay1606