neufieldrobotics/spinnaker_sdk_camera_driver #146: Where to find trigger_msgs package?

maghauke, 2022-02-22:
**System Description**
Camera Name and Model: Blackfly S Gige
Operating System: Ubuntu 18.04 arm64
Spinnaker Version: 2.5.0.80
ROS Version: melodic

***Computer details***
Jetson Xavier AGX

**Do the cameras work with SpinView?**
Yes

**Describe the bug**
Successfully running external triggering, but I want to set the the timestamps of each image to the timestamp of the external trigger-signal, seems to me that by understanding how to use the trigger_msgs package this issue is fixed. But I wonder where the trigger_msgs package is defined and how to use it?

shahvi, 2022-02-22:
I think https://github.com/jpsnir/trigger_msgs should be the right package.

lilrookie76, 2022-11-21:
> 

but that package doesn't include sync_trigger.h, so I meet error again

vik748, 2022-12-23:
It's a message definition. The .h file is created when the messages are compiled.