neufieldrobotics/spinnaker_sdk_camera_driver #154: Failing to Launch node in docker container

KopiousKarp, 2022-09-12:
Hello everyone, I am hoping that you guys can help me out. I have a 2 BFLY-U3-23S6C-C. I have a research robotics project where I need to use these cameras to take photos. Currently the main ROS instance is Ubuntu 22 running ROS Humble Hawksbill. Since this driver is not compatible with that configuration, I want to build a docker container to run it and then I can pipe the photos out from there.

Here is my Dockerfile:
`
# This based off an auto generated Dockerfile for ros:ros-base
# generated from docker_images/create_ros_image.Dockerfile.em
FROM ros:kinetic-ros-base

# install bootstrap tools
RUN apt-get update && apt-get install --no-install-recommends -y \
	apt-utils		\
    build-essential \
    python-rosdep \
    python-rosinstall \
    python-vcstools \
    && rm -rf /var/lib/apt/lists/*

# bootstrap rosdep
RUN useradd -ms /bin/bash sparkslab
RUN rm /etc/ros/rosdep/sources.list.d/20-default.list
RUN rosdep init
RUN runuser -l  sparkslab -c 'rosdep update'
# install ros packages
RUN apt-get update && apt-get install -y --no-install-recommends \
    ros-kinetic-ros-base \
	usbutils		\
	tree			\
	vim				\
	udev	   		\
	wget    		\
	libunwind-dev		\
	ros-kinetic-cv-bridge   \
	ros-kinetic-image-transport

RUN rm -rf /var/lib/apt/lists/*

#getting the Spinnaker SDK installed in the container
#the readme lists these dependencies
RUN sudo apt-get update && apt-get install -y --no-install-recommends \
	libavcodec-ffmpeg56		\
	libavformat-ffmpeg56	\
	libswscale-ffmpeg3		\
    libswresample-ffmpeg1	\
	libavutil-ffmpeg54		\
	libusb-1.0-0		

#get the ROS & SDK driver into the container
COPY ./src/spinnakerSDK /spinnakerSoftware/spinnakerSDK
COPY ./src/install_driver.sh /spinnakerSoftware/install_driver.sh
RUN chmod +x /spinnakerSoftware/install_driver.sh
`    

**System Description**
Camera Name and Model: BFLY-U3-23S6C-C
Operating System:  Docker ROS kinetic image has ubuntu 16 
Spinnaker Version:  2.0.0.146
ROS Version: Kinetic

***Computer details***
Processor: Intel 1165G7
RAM: 32GB

For USB 3.0 cameras:
On same USB 3.0 bus or different buses?
Same Bus
Is the USB 3.0 hub externally powered?
No Hub yet

**Do the cameras work with SpinView?**
Yes on baremetal Ubuntu 22. No in docker, cause I don't want to install it. Correct me if this is a folly.

**Describe the bug**
A clear and concise description of what the bug is.
I load up the Container with all the software that is mentioned by the spinnaker SDK README, then I load all the software for this driver. 
I start up the container and install spinnaker SDK. Then I go through the README for this driver and follow the install and run instructions. 

**Error Messages**
# roslaunch spinnaker_sdk_camera_driver acquisition.launch
[acquisition.launch] is neither a launch file in package [spinnaker_sdk_camera_driver] nor is [spinnaker_sdk_camera_driver] a launch file name
The traceback for the exception was written to the log file

if you can tell me where this log file is I would appreciate it.

Any help is much appreciated! I hope that this benefits others as well.

vik748, 2022-09-12:
This sounds like a very fundamental Ros issue. Could be something as simple as sourcing the right setup.bash or running the command from the correct folder.
I'd suggest making sure you are able to run everything first, maybe in a Virtual Box...

KopiousKarp, 2023-02-01:
Turns out a bunch of stuff was out of whack. Wrong version of ROS, wrong version of spinnaker SDK, wrong version of Ubuntu. I decided to build my own driver. I was woefully ignorant when I made this issue and now I'll just close it up