neufieldrobotics/spinnaker_sdk_camera_driver #159: catkin_make fails due to opencv error

ana-lan, 2023-08-24:
**System Description**
Camera Name and Model: Teledyne FLIR BFS-PGE-27S5C-C
Operating System: Ubuntu 20.04
Spinnaker Version: 64 bit AMD - 3.1.0.79
ROS Version: Noetic

I cloned this repository and tried to build it with `catkin_make` as given in the README file. All the dependencies have been installed as per requirement. I have installed opencv version: 4.2.0
When I try to build this with `catkin_make`, it gives this error:
```
In file included from ~/catkin_ws/src/spinnaker_camera_driver/include/spinnaker_sdk_camera_driver/capture.h:4,
             	from /home/~/catkin_ws/src/spinnaker_camera_driver/src/capture.cpp:1:
~/catkin_ws/src/spinnaker_camera_driver/include/spinnaker_sdk_camera_driver/std_include.h:9:10: fatal error: cv.h: No such file or directory
	9 | #include <cv.h>
  	|      	^~~~~~
compilation terminated.
make[2]: *** [spinnaker_camera_driver/CMakeFiles/acquilib.dir/build.make:63: spinnaker_camera_driver/CMakeFiles/acquilib.dir/src/capture.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 93%] Built target urg_node
In file included from ~/catkin_ws/src/spinnaker_camera_driver/include/spinnaker_sdk_camera_driver/camera.h:4,
             	from ~/catkin_ws/src/spinnaker_camera_driver/src/camera.cpp:1:
~/catkin_ws/src/spinnaker_camera_driver/include/spinnaker_sdk_camera_driver/std_include.h:9:10: fatal error: cv.h: No such file or directory
	9 | #include <cv.h>
  	|      	^~~~~~
compilation terminated.
make[2]: *** [spinnaker_camera_driver/CMakeFiles/acquilib.dir/build.make:76: spinnaker_camera_driver/CMakeFiles/acquilib.dir/src/camera.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2331: spinnaker_camera_driver/CMakeFiles/acquilib.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j8 -l8" failed
```
When I searched for possible solutions for this issue, it was some transition issue of opencv so as per the solutions available I made changes in the std_include.h:
I changed it 
`#include<cv.h` to `#include <opencv2/opencv.hpp`
and 
`#include<cv.h` to `#include <opencv2/core/core.hpp`

In both the cases it gave me this error:
```
~/catkin_ws/src/spinnaker_camera_driver/src/camera.cpp: In member function ‘cv::Mat acquisition::Camera::convert_to_mat(Spinnaker::ImagePtr)’:
~/catkin_ws/src/spinnaker_camera_driver/src/camera.cpp:106:34: error: ‘class Spinnaker::IImage’ has no member named ‘Convert’
  106 |     	convertedImage = pImage->Convert(PixelFormat_BGR8); //, NEAREST_NEIGHBOR);
  	|                              	^~~~~~~
~/catkin_ws/src/spinnaker_camera_driver/src/camera.cpp:108:28: error: ‘class Spinnaker::IImage’ has no member named ‘Convert’
  108 |   convertedImage = pImage->Convert(PixelFormat_Mono8); //, NEAREST_NEIGHBOR);
  	|                        	^~~~~~~
~/catkin_ws/src/spinnaker_camera_driver/src/capture.cpp: In member function ‘void acquisition::Capture::run_soft_trig()’:
~/catkin_ws~/catkin_ws/src/spinnaker_camera_driver/src/capture.cpp:1000:42: error: ‘CV_WINDOW_NORMAL’ was not declared in this scope; did you mean ‘CV_RAND_NORMAL’?
 1000 | 	if (LIVE_)namedWindow("Acquisition", CV_WINDOW_NORMAL | CV_WINDOW_KEEPRATIO);
  	|                                      	^~~~~~~~~~~~~~~~
  	|                                      	CV_RAND_NORMAL
~/catkin_ws/src/spinnaker_camera_driver/src/capture.cpp:1000:61: error: ‘CV_WINDOW_KEEPRATIO’ was not declared in this scope
 1000 | 	if (LIVE_)namedWindow("Acquisition", CV_WINDOW_NORMAL | CV_WINDOW_KEEPRATIO);
  	|                                                         	^~~~~~~~~~~~~~~~~~~
~/catkin_ws/src/spinnaker_camera_driver/src/capture.cpp:1049:23: error: ‘cvWaitKey’ was not declared in this scope
 1049 |         	int key = cvWaitKey(1);
  	|                   	^~~~~~~~~
~/catkin_ws/src/spinnaker_camera_driver/src/capture.cpp:1073:21: error: ‘cvDestroyAllWindows’ was not declared in this scope
 1073 |                 	cvDestroyAllWindows();
  	|                 	^~~~~~~~~~~~~~~~~~~
~/catkin_ws/src/spinnaker_camera_driver/src/capture.cpp:1101:21: error: ‘cvDestroyAllWindows’ was not declared in this scope
 1101 |                 	cvDestroyAllWindows();
  	|                 	^~~~~~~~~~~~~~~~~~~
~/catkin_ws/src/spinnaker_camera_driver/src/camera.cpp: In member function ‘cv::Mat acquisition::Camera::grab_mat_frame()’:
~/catkin_ws/src/spinnaker_camera_driver/src/camera.cpp:100:1: warning: control reaches end of non-void function [-Wreturn-type]
  100 | }
  	| ^
make[2]: *** [spinnaker_camera_driver/CMakeFiles/acquilib.dir/build.make:76: spinnaker_camera_driver/CMakeFiles/acquilib.dir/src/camera.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [spinnaker_camera_driver/CMakeFiles/acquilib.dir/build.make:63: spinnaker_camera_driver/CMakeFiles/acquilib.dir/src/capture.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2331: spinnaker_camera_driver/CMakeFiles/acquilib.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j8 -l8" failed

```

Can someone help me with this issue?

ChunQiuBC, 2023-08-29:
change  
CV_WINDOW_NORMAL to WINDOW_NORMAL  ,
cvWaitKey to WaitKey ,
cvDestroyAllWindows to DestroyAllWindows 
might be helpful，and  I'm still wondering the first error about Convert，If you have any solution，please reply，thx.

rFalque, 2023-10-12:
CV_WINDOW_NORMAL to WINDOW_NORMAL ,
cvWaitKey to waitKey ,
cvDestroyAllWindows to destroyAllWindows

Same looking to solve the Convert, any update on this?

alfred2002, 2024-03-04:
I successfully solve the "Convert" problem by changing my Spinnaker version to a lower one.
The "Convert"  might be an implicit function belongs to pImage, but in version 3.2.0, I cannot find it through searching.
After I installed version 2.2.0, the error disappeared.